kanflo/opendps #149: DPS5015 Overcurrent Protection Error

Schnilz, 2019-06-26:
I just flashed my DPS5015 with MODEL=DPS5015  (I did the suggested dump beforehand [dump.zip](https://github.com/kanflo/opendps/files/3330957/dump.zip) ) . Now it turns on and shows the UI and i can set current / voltage, but as soon as i press the ON button the display flashes i have nothing connected to the terminals....

m-kozlowski, 2019-06-26:
It might me silly, but have you tried to increase current limit before turning output on?
AFAIK it's set to 0 after flash.

Schnilz, 2019-06-26:
now i feel dumb :D
had it set to 20mA as i expected something like that... increasing  it to 1A makes it power on.
It shows there are 3 to 16 mA flowing with nothing connected... is that a calibration error ?
if so how do i do that ?

m-kozlowski, 2019-06-26:
use dpsctl.py calibration routine and follow instructions

closed 2019-06-26 by Schnilz