kanflo/opendps #152: Measure output power and energy

danielkucera, 2019-09-10:
Would it be possible to display output power in Watts and energy in Watthours since startup? I'd like to log battery charging and I'd like to have it quite precise.

Xenoamor, 2019-09-10:
It could be programmed to do so but the issue with measuring watt hours is that you have to integrate the readings. This means you need high frequency sampling but also very accurate measurements to ensure you get reliable results.

Unfortunately the DPS hardware has poor current measurement capabilities so unless you're dealing with fairly high currents it likely wouldn't yield any useful information

danielkucera, 2019-09-10:
There could be a disclaimer about the accuracy. 

My main usecase is charging bateries where the current doesnt change so dramatically. And I wouldn't even need to display it, serial json output is fine. 

Althogh I can programatically mutiply voltage and current readings from serial output, I think I can gain some precision it it was done by the firware itself.

kanflo, 2019-09-10:
It's a nice idea for battery changing as you say.

kanflo, 2019-10-10:
This is being worked on in #166 and will hopefully soon reach master.