kanflo/opendps #153: Voltage is off

danielkucera, 2019-09-10:
Today I've upgraded my DPS5005 to OpenDPS but when I set voltage, the output voltage is off. See following table:

Set,Displayed,Measured
27,25.7,25.59
20,19.04,18.94
15,14.28,14.22
10,9.53,9.48
5,4.78,4.74
3.3,3.17,3.13

I'm attaching original firmware data sets:

[3.7V-off.txt](https://github.com/kanflo/opendps/files/3595976/3.7V-off.txt)
[3.7V-on.txt](https://github.com/kanflo/opendps/files/3595977/3.7V-on.txt)
[5V-off.txt](https://github.com/kanflo/opendps/files/3595978/5V-off.txt)
[5V-on.txt](https://github.com/kanflo/opendps/files/3595979/5V-on.txt)

Is this a bug or do I need to calibrate something?

JackNewman12, 2019-09-10:
You will need to run the calibration script. 

The main commands you will need are:
```
dpsctl.py -C
dpsctl.py -cr
```

Pull request #120 was useful to me

If you are using the USB adapter that came with the DPS you will need to change the baudrate in your build #135 (9600 works for me)

There is a ticket to add this calibration routine information to the readme #131  (This issue also caught me by surprise when I got my DPS a week ago)

I would like to see your calibration parameters and see if they lineup with mine. If so it might be worth recording them somewhere for future users.

Mine:
```
Calibration Report:
	A_ADC_K = 1.67881917953
	A_ADC_C = -117.5
	A_DAC_K = 0.664585292339
	A_DAC_C = 263.0
	V_ADC_K = 13.1308298111
	V_ADC_C = -124.792572021
	V_DAC_K = 0.0761880800128
	V_DAC_C = 3.04274582863
	VIN_ADC_K = 16.7254924774
	VIN_ADC_C = 20.6332607269
```

kanflo, 2019-09-10:
Thanks @JackNewman12

danielkucera, 2019-09-11:
I'm having trouble running it: ```danman@silverhorse:/storage/Projects/opendps/dpsctl$ ./dpsctl.py -d /dev/ttyUSB0 -q
Func       : calibration (off)
  A_DAC    : 2
  V_DAC    : 0
V_in       : 12.21 V
V_out      : 0.00 V
I_out      : 0.000 A
danman@silverhorse:/storage/Projects/opendps/dpsctl$ ./dpsctl.py -d /dev/ttyUSB0 -q
Func       : calibration (off)
  A_DAC    : 2
  V_DAC    : 0
V_in       : 12.22 V
V_out      : 0.00 V
I_out      : 0.000 A
danman@silverhorse:/storage/Projects/opendps/dpsctl$ ./dpsctl.py -d /dev/ttyUSB0 -C
For calibration you will need:
	A multimeter
	A known load capable of handling the required power
	A thick wire for shorting the output of the DPS
	2 stable input voltages

Please ensure nothing is connected to the output of the DPS before starting calibration!

Would you like to proceed? (y/n): y

Input Voltage Calibration:
Please hook up the first lower supply voltage to the DPS now
ensuring that the serial connection is connected after boot
Type input voltage in mV: 12250
Error: protocol error (-3).
danman@silverhorse:/storage/Projects/opendps/dpsctl$ ./dpsctl.py -d /dev/ttyUSB0 -q
Func       : calibration (off)
  A_DAC    : 2
  V_DAC    : 0
V_in       : 12.22 V
V_out      : 0.00 V
I_out      : 0.000 A
```

danielkucera, 2019-09-11:
I switched to 9600bps, I can pass more steps but still I cannot finish the procedure. Please see attached log.

[calibration.txt](https://github.com/kanflo/opendps/files/3602696/calibration.txt)

kanflo, 2019-09-11:
Could you rerun with ```-v``` so we can see the traffic on the UART?

danielkucera, 2019-09-12:
Here it is but I have a bad feeling that something went wrong with the device. The output voltage is 0V no matter what I set :(
[calib2.txt](https://github.com/kanflo/opendps/files/3603893/calib2.txt)

danielkucera, 2019-09-12:
danman@silverhorse:/storage/Projects/opendps/dpsctl$ ./dpsctl.py -d /dev/ttyUSB0 -q -b 9600 -v
Communicating with /dev/ttyUSB0
TX  5 bytes [7e 04 40 84 7f]
RX 48 bytes [7e 84 01 7d 5e 9d 00 00 00 00 01 ff ff ff ff 00 63 76 00 76 6f 6c 74 61 67 65 00 35 30 30 30 00 63 75 72 72 65 6e 74 00 32 30 30 30 00 38 da 7f]

Func       : cv (on)
  current  : 2000
  voltage  : 5000
V_in       : 32.41 V
V_out      : 0.00 V
I_out      : 0.000 A

JackNewman12, 2019-09-12:
Curious what the calibration report `-cr` will show. I expect it might be corrupt. Could you dump that for me?

Can you do a `--calibration_reset` and see if that brings you back to your original behaviour? 

Someone else might like to comment. But I believe you should be able to manually set the voltage output via `dpsctl.py -c VOUT_DAC=XXX` to prove the device is still working.

Enabling the debug calibration plots might be useful to ensure the maths isn't going horrifically wrong:
https://github.com/kanflo/opendps/blob/94cb035bade6d2caeb58ba41b2c935aad3b995fd/dpsctl/dpsctl.py#L51-L53

Have you attempted to apply my calibration parameters and see how it behaves?

danielkucera, 2019-09-12:
```
danman@silverhorse:/storage/Projects/opendps/dpsctl$ ./dpsctl.py -d /dev/ttyUSB0 -b 9600 -v -cr
Communicating with /dev/ttyUSB0
TX  5 bytes [7e 12 32 73 7f]
RX 58 bytes [7e 92 01 00 07 07 86 00 00 00 00 00 00 3f db 43 96 c2 ed 05 1f 3f 26 e9 79 43 90 4e 35 00 00 00 00 00 00 00 00 7d 5f 80 00 00 7d 5f c0 00 00 41 86 16 fb 42 a1 07 c6 b9 73 7f]

Calibration Report:
	A_ADC_K = 1.71300005913
	A_ADC_C = -118.510002136
	A_DAC_K = 0.652000010014
	A_DAC_C = 288.610992432
	V_ADC_K = 0.0
	V_ADC_C = 0.0
	V_DAC_K = inf
	V_DAC_C = nan
	VIN_ADC_K = 16.761220932
	VIN_ADC_C = 80.5151824951
	VIN_ADC = 1926
	VOUT_ADC = 7
	IOUT_ADC = 0
	IOUT_DAC = 0
	VOUT_DAC = 0
```

danielkucera, 2019-09-12:
Phew... I was starting to be a little bit worried, but now it is in original state:
```
danman@silverhorse:/storage/Projects/opendps/dpsctl$ ./dpsctl.py -d /dev/ttyUSB0 -b 9600 -v --calibration_reset
Communicating with /dev/ttyUSB0
TX  5 bytes [7e 14 52 b5 7f]
RX  6 bytes [7e 94 01 c4 0e 7f]

danman@silverhorse:/storage/Projects/opendps/dpsctl$ ./dpsctl.py -d /dev/ttyUSB0 -b 9600 -v -cr
Communicating with /dev/ttyUSB0
TX  5 bytes [7e 12 32 73 7f]
RX 56 bytes [7e 92 01 01 71 07 84 00 46 0d dd 01 6a 3f db 43 96 c2 ed 05 1f 3f 26 e9 79 43 90 4e 35 41 52 9f be c2 c9 80 83 3d 93 74 bc 3f ec cc cd 41 85 f7 cf 42 80 39 58 93 e1 7f]

Calibration Report:
	A_ADC_K = 1.71300005913
	A_ADC_C = -118.510002136
	A_DAC_K = 0.652000010014
	A_DAC_C = 288.610992432
	V_ADC_K = 13.1639995575
	V_ADC_C = -100.750999451
	V_DAC_K = 0.0719999969006
	V_DAC_C = 1.85000002384
	VIN_ADC_K = 16.7460002899
	VIN_ADC_C = 64.1119995117
	VIN_ADC = 1924
	VOUT_ADC = 369
	IOUT_ADC = 70
	IOUT_DAC = 3549
	VOUT_DAC = 362
```

JackNewman12, 2019-09-12:
Interesting how 
```
V_ADC_K = 0.0
V_ADC_C = 0.0
V_DAC_K = inf
V_DAC_C = nan
```
got written to the device. Those calibration plots might actually be helpful in this situation. I guess the maths went funky or the communication got corrupt.

danielkucera, 2019-09-12:
It's strange that during all calibrations the 10% and 90% output never changed. It was always the same value (either 0mV or 4750mV)