kanflo/opendps #174: Dynamic memory usage?

Xenoamor, 2019-10-01:
Any ideas how the following code compiles?
https://github.com/kanflo/opendps/blob/c1a8ab69d97ee3a329a2283ec88c49150537978c/opendps/protocol_handler.c#L348-L351

I've never seen `uint8_t buffer[variable];` compile before

Xenoamor, 2019-10-01:
I've scrubbed this in #176 as I really didn't like the look of it

kanflo, 2019-10-02:
I guess this is one of the places where a static allocation would be suitable.

Xenoamor, 2019-10-02:
As far as I'm aware this isn't valid C at all. It still baffles me how it compiles

m-kozlowski, 2019-10-02:
[I think it is](https://gcc.gnu.org/onlinedocs/gcc/Variable-Length.html)
Well, sort of, because gcc is the only compiller I've used that has it implemented;)

Xenoamor, 2019-10-02:
I had no idea this was a thing. Seems pretty dangerous to me if you're trying to avoid the heap