kanflo/opendps #183: openDPS Cases ♥️

ElectricCookie, 2019-10-03:
Hey this is not an issue but I wanted to share the case I made for my small openDPS build. Maybe others can join and and help build a small gallery of cool custom power supplies
[image]
[image]
[image]

Xenoamor, 2019-10-03:
Damn I like that. I've got the "official" ones at the moment
[image]

kanflo, 2019-10-08:
Cool case! I have a 3D printed minimalistic case that should reach Thingiverse when I get around to adds some finishing touches.

[image]

MartinDejmal, 2019-10-09:
I have a compact one. https://cad.onshape.com/documents/a387dafa34ddb2b0b1734add/w/661b6e36556eeb673311f490/e/dcde09774cb5d0aa43c715d4

Here are some real pictures in author's blog (in czech): http://joysfera.blogspot.com/2017/04/novy-digitalni-laboratorni-zdroj.html

polihedron, 2019-10-16:
Mine is a bit bigger, case is from mini ITX PC, I have 3D printed black frames for display, and adapters for micro-switches buttons. Case, fans, wires and replaced heating sinks are from local store, rest is from china.
[image]
[image]
[image]
[image]

ElectricCookie, 2019-10-19:
That's stunning!

JackNewman12, 2019-10-22:
Small form factor powered from USB-C PD.

I went with USB-C as all of my devices are slowly going that way; Laptop charger, phone, soldering iron (TS80 love it), and now lab supply. 

[image]

Uploaded the 3D files here:
https://www.thingiverse.com/thing:3930817

ElectricCookie, 2019-11-28:
@JackNewman12 why isn't it pulling more than 5V supply voltage?

JackNewman12, 2019-11-28:
Good catch, I had it plugged into my computer for the photo.
Thanks!

TheRealGreatOldOne, 2020-01-06:
I’ve still to update my firmware to open DPS, but here’s my “dual channel” bench supply:
https://www.thingiverse.com/thing:3466963

polihedron, 2020-05-11:
yet another one, in caterpillar or dachshund style.:) 
this time with 3D printed housing, inside is a lot of space because I use common mode choke with big capacitors to filter high and low frequency noise. 
some improvements need to be done, but so far it's working and testing MPPT charge controller.

[image]
[image]
[image]
[image]

Conor-Burns, 2020-07-14:
Absolutey love the custom firmware :D 

I've printed a case to accept DC 24V 5A from a Laptop PSU - the big red switch switches the power and splits it to the left banana terminals and the dps5005. The left terminals are for my TS100 soldering iron and the right ones are for probes connected to the dps.

Then there is a buck converter in there for an esp8266 with the proxy software :)

Thankfully I was able to solder some wires to the new JTAG interface - makes flashing easier when UART decides to fail on me lol.
[image]
[image]
[image]
[image]
[image]

4nthonylin, 2020-10-08:
Just wanted to share my case design here! https://www.prusaprinters.org/prints/42504-dps5005-case
[image]