kanflo/opendps #197: DPH5005 difference/support?

Harvie, 2019-10-25:
There are devices called DPH5005, DS5005 and similar... I am bit confused, they all look the same as DPS5005. Do they differ? Are they supported?

polihedron, 2019-10-25:
nope, I don't think so, DPH series are buck-boost converters, they differ from the step down buck DPS series.

polihedron, 2019-10-25:
DS5005?

Harvie, 2019-10-25:
> Where did you find DS5005?

My bad, there's no such device, it was probably a typo.

> DPH series are buck-boost converters, they differ from the step down buck DPS series.

buck-boost does not seem to be much different from buck converter in terms of software... You just watch some feedback voltage using ADC and produce PWM signal to drive transistor according to it. Do you think it's possible to have support for these devices in future?

[image]

polihedron, 2019-10-25:
dunno, DPH5005 pcb looks like DPS5015 but it's different.
you can replace orginal STM32f1 with new one bought from local store and try to upload new firmware, just in case if something went wrong.

polihedron, 2019-10-25:
#11

polihedron, 2019-10-25:
Compatibly with DPH3205? #11