kanflo/opendps #198: Stabilise and improve UART protocol

kanflo, 2019-10-26:
There are a few design flaws in the current UART upgrade protocol and the use thereof, as mentioned in #191 

1. ```dpsctl.py``` does not do do retransmits
2. As there are no counters in the ```CMD_UPGRADE_DATA```frames, a resent frame might result in ```device reported CRC error``` at the end of the upgrade process.

As this leads to changing the boot loader, the following should be added while at it:

3. There should be protocol support to change the baud rate.
4. Backwards compatibility should be kept (old/new ```dpsctl```should be able to upgrade new/old dpsboot for all permutations of 'old' and 'new').

kanflo, 2019-10-26:
I guess a new ```CMD_UPGRADE_DATA2``` frame is needed with a counter and ```dpsctl``` could start with this one and fall back to ```CMD_UPGRADE_DATA```. The silly thing is that the boot loader does not NAK frames with unknown commands meaning we will need to throw a handful of ```CMD_UPGRADE_DATA2``` at the DPS before falling back the old protocol. Gah!

A boot loader upgrade method should be implemented too for the brave out there.

hseibel12, 2019-10-27:
It is really a good idea to have some kind of handshake dialogue for the bootloader and to repeat a broken frame. I found (#191) that the device just breaks at some point in communication where the GUI is still running. But I am not quite sure if this is always the same behaviour. I will try to verify this tomorrow.

Harvie, 2019-11-22:
BTW has opendps ability to set device address in the serial protocol... I believe that original firmware allows to use multiple devices on single serial line (= they only TX when queried by specific address)

4nthonylin, 2020-09-25:
Any updates on this? I cannot upload firmware due to `Error: device reported CRC error.`