kanflo/opendps #199: DPS5020 Support

lukedempsey, 2019-10-29:
Hi All,

I'm looking at adding support for the DPS5020, but am new to this repo so I thought it may be more efficient if I post my progress here.

I have run the ocd-client.py script with the results attached below.

[M1-on.txt](https://github.com/kanflo/opendps/files/3781763/M1-on.txt)
[M2-on.txt](https://github.com/kanflo/opendps/files/3781764/M2-on.txt)
[ROT-but-on.txt](https://github.com/kanflo/opendps/files/3781765/ROT-but-on.txt)
[SET-on.txt](https://github.com/kanflo/opendps/files/3781766/SET-on.txt)
[5V-10A-off.txt](https://github.com/kanflo/opendps/files/3781767/5V-10A-off.txt)
[5V-off.txt](https://github.com/kanflo/opendps/files/3781768/5V-off.txt)
[5V-on.txt](https://github.com/kanflo/opendps/files/3781769/5V-on.txt)
[10V-off.txt](https://github.com/kanflo/opendps/files/3781770/10V-off.txt)
[10V-on.txt](https://github.com/kanflo/opendps/files/3781771/10V-on.txt)
[button-on.txt](https://github.com/kanflo/opendps/files/3781772/button-on.txt)

I flashed DPS5015 firmware for a starting point as it seems extremely similar to the DPS5020. 
The results are that the fan turns on, but other feedback (screen black). I am more convinced it is my method of flashing it rather than the hardware difference that is failing me. I did flash this a different way to instructed as I am using WSL so this may have injected an error:

- Launched OpenOCD through cmd, as WSL couldn't detect the ST-Link-V2.
In WSL:
```
telnet localhost 4444
reset halt
flash erase_address unlock 0x08000000 0x10000
```
The flashing script can't due to the filepath being different in WSL (described here https://gojimmypi.blogspot.com/2017/04/opendps-with-dps5015.html)
So to resolve this, the elf files were precompiled and manually flashed:
```
program c:/workspace/opendps/opendps/opendps.elf verify reset
program c:/workspace/opendps/opendps/dpsboot/dpsboot.elf verify reset
```

I reran a couple of ocd-client.py tests.
[opendps.txt](https://github.com/kanflo/opendps/files/3781816/opendps.txt)
[opendps_M1.txt](https://github.com/kanflo/opendps/files/3781817/opendps_M1.txt)
[opendps_pwr_on.txt](https://github.com/kanflo/opendps/files/3781818/opendps_pwr_on.txt)

They show the button feedback correctly, but also a lot of generic differences between the stock FW - I'm not sure how this compares to other OpenDPS models' diffs - are there any out there that I can compare to?

polihedron, 2019-10-29:
Hi, have you tried flashing with serial converter? If you flashed dspboot properly, then you should be able to flash bin file. 
With last commits there is an issue with serial flashing, but version from Jun 16, 2019 should be ok.
https://github.com/kanflo/opendps/tree/08331f4c7225a9d6c30c4361cdcc5acf99ff1b74

I run on windows virtualbox with linux mint distribution, no problems with flashing or detection ST-Link-V2 clone, nice and easy.

hseibel12, 2019-10-29:
I also tried to get the WSL running for this job. To get the ST-Link stick running on WSL you need some extra configuration for the serial interface. Later I dropped this idea because I did not found so much advantages using WSL. I have been using Virtualbox + XUbuntu over a long time. It runs well (and also fast enough) for compiling and flashing the DPS. As polihedron stated the firmware-upload is currently broken and you need to use an older release or only use flashing be means of an ST-Link programmer. I am using a cheap model from Amazon - it runs very well on USB.

lukedempsey, 2019-10-29:
Thanks guys, abandoning WSL was a good move - works will in VM.

I'll post a PR soon for the 5020 variant

lukedempsey, 2019-10-30:
Ok, I think im ok with DPS5020 support. PR posted https://github.com/kanflo/opendps/pull/200

closed 2019-10-30 by lukedempsey