kanflo/opendps #204: Future RIDEN RD6006 Support?

MartinDejmal, 2019-11-11:
Has anyone got their hands on RIDEN RD6006? 
https://www.aliexpress.com/item/4000282551930.html

It looks like a bigger successor of DPxNNNN series of DC/DCs. If anyone is interested I have one at home and I am available to do any testing to help OpenDPS development.

polihedron, 2019-11-12:
mcu is STM32F103RCT6.
[image]

Xenoamor, 2019-11-12:
This looks really awesome. Looks much more "professional" and the extra buttons are nice. Saying that though this is definitely a different beast

The CPUs are both M3 cores but this one is 72 MHz as opposed to what I think is the current 24 MHz. This also has 3x ADCs unlike the current 1x ADC so we can sample things at the same time. This would be perfect for an oscilloscope mode and would help to push up our sample rate.

I'd be interested to know if they've fixed #7 which was an inherent design issue with the first generation. My gut feeling is this hasn't been resolved

I imagine the buttons are multiplexed but they could be reached through an i2c extension or something as the previous model had limited GPIOs remaining.

If anyone knows how the actual performance of this device is then I might be pretty interested in getting involved. I think if we're careful we'll be able to maintain backwards compatibility. Am I right in thinking that this is a new bigger display? That could be an issue although it's a great plus from a usability perspective

Looks like this one comes with a thermometer and a fan by default through which is definitely a plus. Looks like the interface is the same though, USB serial or a ESP32 extension board. This time though they look to have a more comprehensive computer application. I imagine we could squeeze a lot of data over that ESP32 link for oscilloscope stuff. Can't say I'm a fan of the USB socket being on the front though.

[There's a more involved video from here on them here](https://www.youtube.com/watch?v=pz0T9rD4Ij0). What looks like a ground terminal on the front is actually some kind of battery charging circuit? Very odd to say the least

I guess the real question is though, what can we bring to this device that isn't already there? Hopefully someone makes a detailed video of it so we can get an idea of what the device is like. There's still no linear regulator front end on this thing which means it will never be that accurate unfortunately and honestly I'm only really interested in the 0-1A range

kanflo, 2019-11-19:
Really cool device, thanks for the info @MartinDejmal. Getting OpenDPS run on that thing would be nice but sandy I do not have the bandwidth to do it.

Xenoamor, 2019-11-24:
[The EEVBlog forums are already discussing it.](https://www.eevblog.com/forum/testgear/ruideng-riden-rd6006-dc-power-supply/) I advise people keep an eye on that thread if they want more information

Xenoamor, 2019-11-27:
And the video we've all been waiting for...
https://www.youtube.com/watch?v=0qjLx_HsKUQ

MartinDejmal, 2019-11-28:
Awww, awesome! I want to start hacking the PSU, but maybe I should buy another sample - I use it as a daily driver and I do not want to mess it ... :D

zoobab, 2019-12-02:
Hackaday article: https://hackaday.com/2019/12/01/turning-a-bad-bench-supply-into-a-better-bench-supply/

jcocovich, 2020-01-19:
Unfortunately for me, the usb port is not recognized. Tried 3 computers and all ports. Asked for a refund.

polihedron, 2020-01-19:
that's good, now you have free sample for developing OpenDPS. :)

nygma2004, 2020-02-24:
Just a quick thought process. I stumbled upon this: https://github.com/Baldanos/rd6006. It says that via the USB port you can communicate with the RD6006 using Modbus over serial. And in the menu, I remember that for communication you can pick USB, Wifi and TTL. What if in TTL mode it accepts Modbus commands from the RX TX pins of the ESP header? That would be easy, there are many Modbus libraries out there, I programmed ESP8266 to talk to Modbus devices over an TTL to RS486 converter.