kanflo/opendps #205: Maybe i bricked my DPS5005

Hansimausi1984, 2019-11-18:
Hi There,

maybe someone can help me :-)

I flashed my DPS5005 with opendps but around 1 sec after i enable the output the screen goes black and all communication (dpsctl and openocd) stops. Also i cant measure any output within this sec. Only reconnecting the Input brings it back on.
If the output stays off, everything works as expected.

One thing i did wrong while flashing, i misread the pins and connected SWDCLK to nRST and SWDIO to VDD and for sure i forgot the "red text" so i don't have a dump of the STM32 peripheral settings....

Any ideas how to bring it back to life? Or better order a new one?

Thanks

Xenoamor, 2019-11-18:
What power supply are you using to power the device? Did the device work before you flashed it?

Hansimausi1984, 2019-11-19:
Hi @Xenoamor ,

its connected to a cheap 36V/7A from ali, i didnt measure the current, but the voltage stays on a constant 36V when i enable the output.
To be honest i dont know if the output worked the moment before i flashed it. I bought the DPS ~1 year ago, used it one time with a cheap notebook power supply and moved it into my "electronic projects you need to finish"-box.
I ll try to disassemble the DPS this weekend, maybe i blew up something when i connected the STLink ...

Regards

polihedron, 2019-11-19:
Hi, can you check calibration report? maybe there is something wrong with DAC values?
- replace power supply if you have any other, 
- reflash fw, 
- check for any short circuit at output, and measure resistance when DPS is off, 
- check or replace output capacitors.

Hansimausi1984, 2019-11-19:
Hi,
thanks for the guidance.
Already reflashed some times without any difference. Also tried to build it using the dockerfile.
The other points i ll try to check on the weekend.

Regards

Hansimausi1984, 2019-11-22:
Hi,

lets close this issue ...
Today i wanted to take a look again.
After connecting another power supply and enabling the output "something" in the lower left where the xl7005 is located started to smoke and now the whole DPS is dead :-)
Everything in this area looks okay and i think even if i find/replace the damaged part i would never rely on this DPS again.
Already ordered a new one.

Thanks for your help and input

closed 2019-11-22 by Hansimausi1984