kanflo/opendps #206: [question] Original modbus serial protocol?

Harvie, 2019-11-22:
Does the OpenDPS support original serial protocol provided by original firmware?
Can you please add this option to support the legacy protocol?

So when i have multiple devices and some of them have OpenDPS and some of them don't (because they're not supported, eg. #197) i can use all of them with the same protocol without having to sacrifice the possibility to flash OpenDPS?

kanflo, 2020-07-12:
I could if someone raised a PR about it, I will most likely never get around to do it ;)

closed 2020-07-12 by kanflo

shaneyake, 2024-07-04:
I would also like to see Modbus support, I have multiple units on one RS485 bus, would like to move over to OpenDPS but would need to change all my hardware and software layout.

Thanks