kanflo/opendps #207: Compile error "nonconstant expression for length"

yashm, 2019-12-10:
I get the following compile-time error:

  **CC      font-meter_large.c
  CC      gfx-cc.c
  CC      gfx-crosshair.c
  CC      gfx-cv.c
  CC      gfx-cl.c
  CC      gfx-padlock.c
  CC      gfx-thermometer.c
  CC      gfx-wifi.c
  CC      opendps.c
opendps.c: In function 'ui_handle_event':
opendps.c:552:9: warning: unknown option after '#pragma GCC diagnostic' kind [-Wpragmas]
 #pragma GCC diagnostic ignored "-Wimplicit-fallthrough"
         ^
  CC      gfx-poweron.c
  CC      gfx-poweroff.c
  CC      func_cc.c
  CC      func_cl.c
  CC      func_gen.c
  CC      uui_icon.c
  CC      gfx-square.c
  CC      gfx-saw.c
  CC      gfx-sin.c
  CC      uframe.c
  CC      protocol.c
  CC      protocol_handler.c
  LD      opendps_DPS5005.elf
stm32f100_app.ld:15: nonconstant expression for length
collect2: error: ld returned 1 exit status
make: *** [opendps_DPS5005.elf] Error 1**

Any pointers on how to solve this?

Xenoamor, 2019-12-10:
What version of arm GCC are you using? I recall seeing this error but it was around 5 years ago

yashm, 2019-12-10:
yeah i just solved it - it was an old gcc version

closed 2019-12-10 by yashm