kanflo/opendps #208: Color splash logo at boot

yashm, 2019-12-11:
Is there a way to create a color logo for the boot screen?
Tried making a color PNG in gimp indexed to 3 colors and created a .h file [using this generator](http://www.majer.ch/lcd/adf_bitmap.php).
It looks garbled, even with correct height and width macros.

Has anyone had luck generating custom splash screens?

closed 2019-12-11 by yashm

Xenoamor, 2019-12-11:
Is there a reason you closed this? I can guide you through using our own generator to create the logo if you need. The display uses a special 565 colour format so you can't use that tool

kanflo, 2019-12-11:
The already generated graphics contains some hints on how to generate them:

```
/** Gfx generated from `./gen_lookup.py -i gfx/png/padlock.png -o padlock` */
```