kanflo/opendps #209: CLI mode HOW TO documentation

yashm, 2019-12-11:
In the file dps-model.h you mention:
`you can see real ADC value in CLI mode's stat:`

Is the documentation on how to use it available anywhere specifically for this firmware?

yashm, 2019-12-11:
@kanflo I am happy to tune the parameters for the DPS8005 and send in a PR once I know how to do the above...

Xenoamor, 2019-12-11:
I assume this is referring to the `dpsctrl -cr` command

yashm, 2019-12-11:
The line I'm referring to is this one: https://github.com/kanflo/opendps/blob/c1a8ab69d97ee3a329a2283ec88c49150537978c/opendps/dps-model.h#L18

Does the command you shared show the raw ADC values?

Xenoamor, 2019-12-11:
Yes, the calibration screen does. I'm away at the moment but I believe you press both the up and down arrow at the same time to get there (it's intentionally awkward)

yashm, 2019-12-11:
Oh that was easier than I thought! Is there any documentation for this?

kanflo, 2019-12-11:
Weeeell, the documentation has somewhat fallen behind. Some of the information from my old blog posts are stil valid and some is obsolete. Recompiling them into a GitHub.io page has been on the todo list for quite some time. As always, contributions are welcome ;)

yashm, 2019-12-11:
Can you share how you execute the following?
`over openocd console:  mww 0x40007408 77`

Do we need the FTDI connected for this?
Or just open an openocd console in one window and in another terminal type the above command?

Does the DPS need to be in any particular state for this to work?

kanflo, 2019-12-16:
It seems you found the correct way to execute the openpcd commands. It is strange you don't see any difference when changing ```0x40007408```. Can you try changing output value in the UI and see what happens to that, and other registers?