kanflo/opendps #210: DP50V5A as CC LED driver

yashm, 2019-12-20:
As the title suggests, is there a way to use the DP50V5A as a constant current source LED driver?

In its current state, when I set the voltage to about 7V and set the current limit to 800mA, the DPS flashed to denote CC mode and turns off the output.

I need the DPS to maintain the CC mode and supply the constant current specified.

yashm, 2019-12-20:
Here's a gif of the behavior:
[image]

Xenoamor, 2019-12-20:
It's been a while since I've used this but I believe you hold the set button and rotate the dial left. This should change CV to CC meaning it is in constant current mode

yashm, 2019-12-21:
What is CL mode?

SPS1982, 2023-08-19:
question DP50V5A cannot be calibrated in any way, it incorrectly shows both the output voltage and amperage. I tried to calibrate with different resistances ... where to dig? maybe the memory does not save the calibration?