kanflo/opendps #214: Skip parts of calibration

buffedelic, 2020-04-12:
I am struggling with the task of getting my current reading kind of correct. Voltage is spot on even with load up to 2A in rages of 1-5v. 
Can I get the calibration variables stored and skip the first part of calibration involving Vin and Vout?

polihedron, 2020-05-11:
Hi, somehow for me more convinent and accurate is manual calibration,
check commands like:
./dpsctl.py -d /dev/ttyUSB0 --screen settings

./dpsctl.py -d /dev/ttyUSB0 -c A_ADC_K=your calibration factor

./dpsctl.py -d /dev/ttyUSB0 -c A_ADC_C=your calibration factor

./dpsctl.py -d /dev/ttyUSB0 -p A_DAC=100

./dpsctl.py -d /dev/ttyUSB0 --screen main

or check manual for the script.

Xenoamor, 2020-05-14:
I think -cr might give you all the calibration values as well, not sure