kanflo/opendps #216: DPS5015 output changes from set value once enabled

ilium007, 2020-05-27:
I have compiled for the DPS5015 but output values are inaccurate - see photos.

I set 3.300V and measure 2.858V

I set 15.000V and measure 14.25V

Is there anything I can do to fix or have I just created a paperweight?

[image]

[image]

[image]

[image]

ilium007, 2020-05-27:
I also have a spare v2.4 board that I haven’t touched as yet. 

[image]

ilium007, 2020-05-27:
It turns out my spare board was a DPS5020

[image]

But it’s output measured on the same meter is spot on. Not game now to put OpenDPS on the DPS5020 and end up with two paperweights. 

[image]

[image]

Xenoamor, 2020-05-27:
You need to calibrate the device

When you turn it on the screen displays what the ADCs see after it's internal calibration parameters are applied

If you turn it on and the true output doesn't match what you set it to then the DAC calibration parameters are wrong

ilium007, 2020-05-27:
Is there a guide for this process?

Xenoamor, 2020-05-27:
#120 and #122 are the related PRs that added in the calibration routine. It's done via the communication interface though. There's not a good guide though and I've since moved on from this project

If you don't have a communication interface you can manually set the parameters by pressing m1 + m2 at the same time to get to the calibration screen.. I think anyway

ilium007, 2020-05-27:
Ok thanks. I think I’ll just bin the OpenDPS 5015 and leave my 5020 stock because it just works.

ilium007, 2020-05-27:
I’ve just had a look through the commits referenced above and it looks like I understand enough to calibrate. How far apart do the voltage references have to be? Would 3.3v and 24v be far enough apart?

Xenoamor, 2020-05-27:
Fair enough, they are almost disposable at the cost they are.

I think the main readme.md could do with a disclaimer, pulling the factory programmed calibration parameters is present in the part 3 of the blog but it's not obvious to people who aren't clicking through to ever link

> Would 3.3v and 24v be far enough apart?
Are you referring to the input voltage, the one that feeds into the back of the supply? If so 3.3v sounds kind of low and I'm not sure it will function correctly with it. I used another DPS to feed my one and used 6v as my lower value and then the max I could output. You'll want to set the current limit to be high

ilium007, 2020-05-27:
I did save the txt files during my initial flashing ad per the readme. I didn’t realise there were calibration values in there.

ilium007, 2020-05-27:
When I try to run dpsctl.py within a clean pipenv that has pycrc and pyserial I get the error below

[image]

Both are installed:

[image]

Xenoamor, 2020-05-27:
> I did save the txt files during my initial flashing ad per the readme. I didn’t realise there were calibration values in there.

I've never done this but maybe I recall reading something about it. I just threw on OpenDPS straight away and realised it needed calibrating which is why I added the routine for it

I thought the bug you were seeing was likely fixed in #213 but I guess not. The issue tracker is #215 
I can probably look into this when I'm home if you don't manage to resolve it. We don't use pycrc that extensively so if it's just a case of updating the API calls to the latest version then it's an easy fix

ilium007, 2020-05-28:
Got a bit further tonight, I will try and fix and raise a PR.

```
~/opendps/dpsctl master*
dpsctl-JB5bKZwA ❯ ./dpsctl.py -d /dev/cu.usbserial-1430 -C
For calibration you will need:
	A multimeter
	A known load capable of handling the required power
	A thick wire for shorting the output of the DPS
	2 stable input voltages

Please ensure nothing is connected to the output of the DPS before starting calibration!

Traceback (most recent call last):
  File "./dpsctl.py", line 1204, in <module>
    main()
  File "./dpsctl.py", line 1198, in main
    handle_commands(args)
  File "./dpsctl.py", line 577, in handle_commands
    do_calibration(comms, args)
  File "./dpsctl.py", line 723, in do_calibration
    t = raw_input("Would you like to proceed? (y/n): ")
NameError: name 'raw_input' is not defined

~/opendps/dpsctl master*
dpsctl-JB5bKZwA ❯
```

Xenoamor, 2020-05-28:
Looks like `raw_input` should have changed to `input` when the code was ported to python3

ilium007, 2020-05-28:
Thats it. I just made the change on mine

ilium007, 2020-05-28:
Raised PR:
https://github.com/kanflo/opendps/pull/217

ilium007, 2020-05-28:
Why is the calibration routine asking for a load resistance if the latest calibration uses a solid short? Is it expecting me to input 0?

`Load resistance in ohms:`

ilium007, 2020-05-28:
```
opendps_i7-0FvMkaG2 ❯ ./dpsctl.py -d /dev/cu.usbserial-1430 -C
For calibration you will need:
	A multimeter
	A known load capable of handling the required power
	A thick wire for shorting the output of the DPS
	2 stable input voltages

Please ensure nothing is connected to the output of the DPS before starting calibration!

Would you like to proceed? (y/n): y

Input Voltage Calibration:
Please hook up the first lower supply voltage to the DPS now
ensuring that the serial connection is connected after boot
Type input voltage in mV: 19990

Please hook up the second higher supply voltage to the DPS now
ensuring that the serial connection is connected after boot
Type input voltage in mV: 19990

Output Voltage Calibration:
Finding maximum output V_DAC value.......................... Done

Calibration Point 1 of 2, 10% of Max
Type measured voltage on output in mV: 17230

Calibration Point 1 of 2, 90% of Max
Type measured voltage on output in mV: 15940

Output Current Calibration:
Max output current of your DPS (e.g 5 for the DPS5005) in amps: 5
Load resistance in ohms: 0
Load wattage rating in watts: 0
Please connect the load to the output of the DPS, then press enter
```

ilium007, 2020-05-28:
Backed out of calibration. I had my 12AWG 'thick wire' ready for the current calibration but then it asked for load resistance settings.

Xenoamor, 2020-05-28:
You need a known load for current calibration yes. Ideally a beefy wire wound low resistance one.
If you have thick wire you could use a long length of it I suppose

ilium007, 2020-05-28:
“A thick wire for shorting the output of the DPS”

“shorting”

This to me did not sound like a power resistor. Didn’t I read in of the threads on here that the outputs are literally “shorted”?

Xenoamor, 2020-05-28:
"A known load capable of handling the required power" is the section I'm referring to
At one stage of the calibration you do literally short the terminals though yes

ilium007, 2020-05-28:
Ok. That makes it clearer

ilium007, 2020-05-29:
What defines “required power”? If I have a 24V power supply capable of 10A, let say I want max 20V from the DPS5015 what sort of power resistor am I looking for? The biggest wire wound resistors I can source locally are 10W

polihedron, 2020-05-31:
for 10A current at 20V you will need 2 Ohm 200W resistor, but for short time, like few seconds measure probably 50W will be enough.  It's easer to connect few resistors in paraller or in series, that will reduce needed Wats, per single resistor.
Then you will need combination than will give summary resistance 2 Ohm, for example eight 16 Ohm 10 Wats resistors in pararell give about 2 Ohm summary resistance and about 80Wats.

ilium007, 2020-11-12:
@Xenoamor sorry to dig this up again but my DPS5020 is still bricked and I need to get it working again. You mentioned above that I need a 'thick wire' for shortign the terminals for one step of the calibration routine and you said also that I do need a load resistor. I went and read #122 where you state that the load resistor is no longer required:

> I've tweaked how the constant current calibration works. It no longer uses a load resistor and instead you short the output terminals. This is effectively the equivalent of the open circuit calibration routine for the voltage but for current
> 
> This means it now can sweep the entire current range giving a greater degree of accuracy
> output current sweep shorted

I was about to spend $50 on a 2Ω 200W load resistor but now I'm confused. Do I need the load resistor or can I short the terminals during calibration?

Xenoamor, 2020-11-17:
Sorry for the long response, hopefully it finds you in time. You don't need to buy a load resistor, just short the terminals together. I used a metal ruler but any thick wire should do the trick.
It's been a while but hopefully the python tools explain this when you do the calibration routine

ilium007, 2020-11-18:
ahhh ok, looks like mine will be a brick for a while until I can find some bulk resistance