kanflo/opendps #219: DP50V2A

werdegars, 2020-06-12:
i have DP50V2A
and successfully flashing it
everything was good

but when i tried to turn on the output, 
result no voltage 

even if i changed the voltage and the current value

any idea what is wrong here?

by the way

i have modified **Makefile**
MODEL := DP50V2A

i have added value in **dps-mode.h**
i just copied it from value of the DP50V5A

`#elif defined(DP50V2A)

 #ifndef CONFIG_DPS_MAX_CURRENT

  #define CONFIG_DPS_MAX_CURRENT (2000)

 #endif

 #define ADC_CHA_IOUT_GOLDEN_VALUE  (0x45)

 #define A_DAC_K (float)0.6402f

 #define A_DAC_C (float)299.5518f

 #define A_ADC_K (float)1.74096f

 #define A_ADC_C (float)-121.3943805f

 #define V_DAC_K (float)0.07544f

 #define V_DAC_C (float)2.1563f

 #define V_ADC_K (float)13.253f

 #define V_ADC_C (float)-103.105f

#elif defined(DPS3005)
...`

i have modified also value in **hw.h**

#ifdef DP50V2A
	#define ADC_CHA_VOUT  (4)
#elif
	#define ADC_CHA_VOUT  (9)
#endif
`
noticed i have changed

`#define ADC_CHA_VOUT  (9)  to (4)`

the reason is because if i set to (9)
the displayed voltage output is 0

polihedron, 2020-06-14:
Hi, if you have dump of the peripheral settings before flash fw, that probably will helps a lot.
https://johan.kanflo.com/upgrading-your-dps5005/

You set ADC_CHA_VOUT (4) because on the DP50V2A measuring Vout is at pin 14 (PA4) instead of 19 (PB1)? 
After that you see on display non zero voltage, and multimeter shows output voltage?
I mean it's fixed now?

werdegars, 2020-06-14:
> Hi, if you have dump of the peripheral settings before flash fw, that probably will helps a lot.
> https://johan.kanflo.com/upgrading-your-dps5005/
>
> You set ADC_CHA_VOUT (4) because on the DP50V2A [...]

that is what confuse me, the voltage reading is displaying but no voltage output,
wierd

"if you have dump of the peripheral settings before flash fw"
yeah but i dont understand how to read the informations in it

werdegars, 2020-06-14:
[5V-on.txt](https://github.com/kanflo/opendps/files/4776417/5V-on.txt)

that is the file genereted by using
./ocd-client.py all > 5V-on.txt

polihedron, 2020-06-14:
that's strange, I think you need to match connections on PCB to code, because port PA7 (ADC1_IN7) measures Vout, PB0 (ADC1_IN8) measures Vin, PB1 (ADC1_IN9) measures Iout, PA4 and PA5 are DAC1_OUT and DAC2_OUT they control the output voltage via a TL594 Pulse-Width-Modulation Control Circuit, try to match pin numbers correctly.

If you will share dump files, then maybe someone help you. You can also read ocd-client.py script,
at the bottom there is some comment/description about DAC registers, and hw.c file:

/**
  * @brief Set the output voltage DAC value
  * @param v_dac the value to set to
  * @retval none
  */
void hw_set_voltage_dac(uint16_t v_dac)
{
    DAC_DHR12R1 = v_dac;
}

/**
  * @brief Set the output current DAC value
  * @param i_dac the value to set to
  * @retval none
  */
void hw_set_current_dac(uint16_t i_dac)
{
    DAC_DHR12R2 = i_dac;
}

so, DAC_DHR12R1 is probably the register you're looking for, use command via openocd:
mww 0x40007408 0x00000186
or
mww 0x40007424 0x3cd01860

don't remember with one was working for me.

werdegars, 2020-06-15:
i have managed to get the voltage output **BUT**
if i tried this manually at POWER OFF state

`./ocd-client.py w 0x40007408 169`
and 
`./ocd-client.py w 0x40007414 1d6`

but if at POWER ON state and do that, no output voltage
confusing

closed 2020-06-15 by werdegars

werdegars, 2020-06-17:
i have managed to make it work
thank you for the tips

Funili, 2020-10-09:
Hey werdegars,
can you provide me with your BIN file?
This project is my first contact with the STM32 and I just don't get the program compiled...
Otherwise I only program EPS's
When I flash the bin file "opendps_DPS5005", no voltage is displayed and not switched power on.
Greetings Mazi