kanflo/opendps #221: Issues when runnung dpsctl.py (trying to calibrate a DPS3215)

abzman, 2020-07-03:
I got the initial firmware flashed and it all seems to work, but it intermittently locks up (I never get more than a minute of runtime out of the device before I have to power cycle it).  I flashed it with a firmware compiled with no parameters so that didn't shock me.  I have been trying to calibrate and running into other issues, possibly ones with the dpsctl.py code itself.  First it wouldn't accept my 'y' to calibrate, so I commented that line out and just jumped into punching in the voltage readings.  Then it wouldn't take my pressing of enter at input("Please connect the load to the output of the DPS, then press enter") so I commented that line out and got the load installed first and it went on its merry way.  Now it wants me to short it out but I don't think there's a way for me to comment that out without timing the part swap very carefully, so I think now I need to figure out the real issue here.  Here's the last trace: 

Constant Current Calibration:
Please short the output of the DPS with a thick wire capable of carrying 15.0A, then press enter
Traceback (most recent call last):
  File "./dpsctl.py", line 1204, in <module>
    main()
  File "./dpsctl.py", line 1198, in main
    handle_commands(args)
  File "./dpsctl.py", line 577, in handle_commands
    do_calibration(comms, args)
  File "./dpsctl.py", line 963, in do_calibration
    input("Please short the output of the DPS with a thick wire capable of carrying {}A, then press enter".format(max_dps_current))
  File "<string>", line 0
    
    ^
SyntaxError: unexpected EOF while parsing

Some further questions:
is it better to have a bigger load resistor if possible?
is it needed to have a supply that can give the full amperage I want to have the supply capable of during the calibration?
can I power off the DPS between the low and high voltage measurements if I don't have a high amperage variable supply?

Xenoamor, 2020-07-04:
Are you running this under python 3?
I'd expect this error under python 2

abzman, 2020-07-11:
That was the main issue, I'd suggest changing the first line of the file to specify python3, there are plenty of systems like mine that have both

closed 2020-07-11 by abzman

Xenoamor, 2020-07-11:
> That was the main issue, I'd suggest changing the first line of the file to specify python3, there are plenty of systems like mine that have both

Is there a normal way of doing this? Or are you referring to a comment maybe?

abzman, 2020-07-11:
I just added a 3 to the first line so it reads "#!/usr/bin/env python3" I think whenever someone has just 'python' there it either usually or always resolves to either 'python2' or 'python3' depending on what they use more often, if the code only runs in python3 I would suggest editing the first line to force that, it shouldn't break for anyone I don't think.  Take this with a grain of salt, this isn't my speciality.

kanflo, 2020-07-12:
I need to look into how "modern" systems treat `python3`. IIRC Ubuntu 20.04 ditched python2 meaning `python` starts Python 3 but hopefully there is a `python3` symlink for backwards compatibility.