kanflo/opendps #222: DPS3215 locks up periodically with opendps firmare

abzman, 2020-07-11:
I experienced this before calibration, and it's still happening after calibration.  It's possible the screen just stops responding, because I can hear the fan come on and off when I press the power button.  I have a 36v 11A supply to back this unit, I calibrated it and told it the max amperage it could drive was 10A, this is the resulting calibration report. 

Calibration Report:
	A_ADC_K = 4.377185344696045
	A_ADC_C = -260.4835205078125
	A_DAC_K = 0.25430262088775635
	A_DAC_C = 263.3804931640625
	V_ADC_K = 8.133505821228027
	V_ADC_C = -109.9773178100586
	V_DAC_K = 0.1118573322892189
	V_DAC_C = 8.37867546081543
	VIN_ADC_K = 16.66666603088379
	VIN_ADC_C = 118.33333587646484
	VIN_ADC = 2305
	VOUT_ADC = 14
	IOUT_ADC = 0
	IOUT_DAC = 0
	VOUT_DAC = 0

I have attached the files grabbed before I changed the firmware in case that might help.  

[5v-on.txt](https://github.com/kanflo/opendps/files/4906290/5v-on.txt)
[7v-off.txt](https://github.com/kanflo/opendps/files/4906291/7v-off.txt)
[7v-on.txt](https://github.com/kanflo/opendps/files/4906292/7v-on.txt)
[10v-off.txt](https://github.com/kanflo/opendps/files/4906293/10v-off.txt)
[10v-on.txt](https://github.com/kanflo/opendps/files/4906294/10v-on.txt)
[5v-off.txt](https://github.com/kanflo/opendps/files/4906295/5v-off.txt)

polihedron, 2020-07-11:
hi, maybe it's related to #203

abzman, 2020-07-11:
I lopped out the ifdef around:

#ifdef DPS5015 
#define TFT_CSN_PORT GPIOA
#define TFT_CSN_PIN  GPIO8
#endif

and recompiled.  I had some issues and had to rewipe the whole unit and reflash from scratch, but it seems to work now.  I'll keep testing

abzman, 2020-07-17:
seems stable now, just treat the DPS3215 like a DPS5015

closed 2020-07-17 by abzman