kanflo/opendps #223: DPS5005 Jittery Values

4nthonylin, 2020-09-25:
I noticed that the power supply values for input, output voltage, and current are very jittery. Was wondering if this was expected behavior. I have attached gifs of the input and output voltage.

Input Voltage:
[image]

Output Voltage:
[image]

kanflo, 2021-03-14:
Mine are slightly jittery too. Not sure if it is related to the OpenDPS firmware or the hardware though.

4nthonylin, 2021-03-14:
Probably hardware from the noise of the switching supply. I added an exponential decay filtering algorithm which has improved the jitter.

kanflo, 2021-04-15:
Cool, anything you would like to include in a PR?

4nthonylin, 2021-04-15:
Unfortunately I lost my original changes when I re-imaged my partition. I'll try to recover it.

sercona, 2022-03-15:
has anyone asked for (or thought of) showing the 'admin' value instead of the 'oper' value, for the volts and amps numeric values?  (sorry to borrow snmp terms, lol).  meaning, the set-values, and those are not jittery.

maybe I'm not the typical user, but to me, when I set values on a psu, I dont really care to see ON THE PSU what the circuit is seeing.  if I want to see that, I'll probe the circuit for real and properly (locally with care) and not at the psu's terms.

so, I'd rather see the set-values when the power mode is on as well as off.  I think the admin values are used when power is off, so the logic is just to continue to show that behavior.

I could also see a color-code way to show values and the user would be able to know, immediately, if the display value is admin or oper based.  (I was thinking of just rending the numbers in a different color when the power mode is off vs on).

anyone like the idea of showing the set-values, even when output is 'on'?

no jitter, that way, so its a side benefit.

howie-j, 2022-03-26:
I agree, i would also prefer to see the set values.

kanflo, 2022-04-11:
I prefer the jittery values a I get a sense of the output on the terminals. I agree it is a bit of a nuisance though. If anyone whips up a PR something along the lines of `make JITTERY_VALUES=0` I would be happy to approve.