kanflo/opendps #225: DPS5020 seconds output lag

DatanoiseTV, 2020-11-09:
When I set my 5020 to 10V, it will only change around 15s later. All functions, including function generator seem delayed.
If the function generator is started, turned off 5s later, the function generator output will appear only 10s later on my scope.
Also, the voltages are a bit off (more than stock firmware).

4nthonylin, 2020-11-11:
So I also a delay with the ux and other things. I believe it's due to the high refresh rate and interrupt handler. It seems like we build up events in the event queue which then causes the latency. I made some changes locally to change the interrupt handler and change how the screen is updated and it seems to be working fine now. You can take a look at this patch and try it to see if it fixes the issue:

```diff
diff --git a/opendps/opendps.c b/opendps/opendps.c
index ad1d4b1..99d55d0 100644
--- a/opendps/opendps.c
+++ b/opendps/opendps.c
@@ -88,7 +90,7 @@
 #endif // CONFIG_SPLASH_SCREEN
 
 /** How ofter we update the measurements in the UI (ms) */
-#define UI_UPDATE_INTERVAL_MS  (250)
+#define UI_UPDATE_INTERVAL_MS  (50)
 
 /** Timeout for waiting for wifi connction (ms) */
 #define WIFI_CONNECT_TIMEOUT  (10000)
@@ -518,6 +520,7 @@ static void ui_handle_event(event_t event, uint8_t data)
 #endif // CONFIG_OCP_DEBUGGING
                 ui_flash(); /** @todo When OCP kicks in, show last I_out on screen */
                 opendps_update_power_status(false);
             }
             break;
         case event_ovp:
@@ -532,21 +535,41 @@ static void ui_handle_event(event_t event, uint8_t data)
 #endif // CONFIG_OVP_DEBUGGING
                 ui_flash(); /** @todo When OVP kicks in, show last V_out on screen */
                 opendps_update_power_status(false);
             }
             break;
         case event_button_m1_and_m2:;
             uint8_t target_screen_id = current_ui == &func_ui ? SETTINGS_UI_ID : FUNC_UI_ID; /** Change between the settings and functional screen */
             opendps_change_screen(target_screen_id);
             break;
 
         case event_button_enable:
             write_past_settings();
+            /** Deliberate fallthrough */
+        case event_button_m1:
+        case event_button_m2:
+        case event_button_sel:
+        case event_button_sel_m1:
+        case event_button_sel_m2:
+        case event_rot_left_m1:
+        case event_rot_right_m1:
+        case event_rot_left_m2:
+        case event_rot_right_m2:
+        case event_rot_left_down:
+        case event_rot_right_down:
+        case event_rot_press:
+        case event_rot_left_set:
+        case event_rot_right_set:
+        case event_rot_left:
+        case event_rot_right:
+            uui_refresh(current_ui, false);
+            break;
         default:
             break;
     }

     uui_handle_screen_event(current_ui, event, data);
-    uui_refresh(current_ui, false);
 }
 
 /**
@@ -609,6 +632,13 @@ static void ui_tick(void)
     static uint64_t last_tft_flash = 0;
     static uint64_t last_lock_flash = 0;
 
+    static uint64_t last = 0;
+    /** Update on the first call and every UI_UPDATE_INTERVAL_MS ms */
+    if (last > 0 && get_ticks() - last < UI_UPDATE_INTERVAL_MS) {
+        return;
+    }
+
+    last = get_ticks();
     uui_tick(current_ui);
     uui_tick(&main_ui);
 
@@ -958,8 +988,6 @@ static void event_handler(void)
                     break;
             }
             ui_handle_event(event, data);
-            // update UI immediately on event
-            ui_tick();
         }
     }
 }
```

kanflo, 2021-03-14:
This is very strange. Do you see the delay immediately after power on or after some time of usage?

4nthonylin, 2021-03-14:
So it's been a while but if I remember correctly, on power up there's a slight delay (maybe a few seconds or less?) for this behavior. Haven't seen it since the patch I made.