kanflo/opendps #228: Blank screen on DPS3005

frederikvs, 2021-02-13:
I've recently ordered a new DPS3005, and I'm trying to flash openDPS. However, all I get is a blank screen. Backlight turns on, but nothing is shown.

Using GDB I've been able to determine that openDPS is running correctly (e.g. a breakpoint in the event_handler gets triggered when I push one of the buttons, and the event variable matches the button I pressed).

I've even been able to get 5V out of it by pushing the buttons "in the blind", so I assume it is only the display code that's not working.

Originally I was working with arm-none-eabi-gcc 7.3.1 (which I already had installed). I noticed issue #132, which indicates a black screen with gcc 7.3.0, and mentions that the docker script does work. However, in my case that didn't solve anything.
The docker image seems to be running gcc 7.4.0, but the Dockerfile description mentions 7.1.0, so I modified the docker file to install that version. Also no success.

PCB version is 2.3

Any suggestions on how to proceed?

kanflo, 2021-02-14:
Hmm, it could be either that newer models have a new display requiring a new driver or that the display moved to another SPI bus (seems unlikely though). Unless you can find a fellow '3005 owner who can help you out with a known working binary I guess we are back to reverse engineering these things.

frederikvs, 2021-02-15:
I had a look at a 50V5A I have that's already running openDPS. It looks like it's the same PCB (also marked as v2.3), but with a few different components. So I would think that the actual SPI pins in use are still the same.

Display looks identical, but of course that's no guarantee that the internals are the same. If there's a different chip in there somewhere that needs a few different commands, that might explain what we're seeing.

Would the output of the ocd-client.py help us in figuring that out? Or would we need to snoop the actual SPI going to the display?
I could order a new 3005 from the same place and try to collect the necessary data.

kanflo, 2021-02-15:
cod-client.py would only tell us about the SPI bus used. Snooping would be required to figure out how to talk to this display.

daphreak66, 2021-02-20:
I've closed issue #231 since it seems to be the same problem.

You said the components differ between the working and non-working DPS. Could you provide some pictures? Maybe it's at least possible to identify new and old devices without flashing. I've ordered a new DPS from Amazon and could run ocd-client.py on it, but I have no idea wether it's on old or new device. I ordered the non-working one from aliexpress, and could order another from the same vendor, but it would take a few weeks to get here.

frederikvs, 2021-02-20:
The differences I've noticed between my 2 devices seem to be related to their different voltage ratings. The large capacitors have different values, and the 50V5A has an IC where the 3005 doesn't (backside, between the heatsink and the capacitors). The IC is marked 4264E, which appears to be a power mosfet. There may be other small differences, I haven't investigated that deeply.

[DPS50V5A.zip](https://github.com/kanflo/opendps/files/6014862/DPS50V5A.zip)
[DPS3005.zip](https://github.com/kanflo/opendps/files/6014863/DPS3005.zip)

However, most of the logic seems to live on the front PBC, hidden below the display. I've tried, but haven't had much luck identifying the components hidden there.

I have another device on its way, from the same Aliexpress supplier. As soon as it arrives, I'll share the output of ocd-client here. Perhaps it's something as simple as a different baud rate or something.
If it's more complex, I'm willing to try and hook up a logic analyzer. I'm just a bit worried that the pins to the display are very hard to reach without unsoldering, and in order to unsolder the display I think I'd need to unsolder the back PCB first. Not looking forward to that, but I guess we'll burn that bridge when we get to it :-)

daphreak66, 2021-02-21:
Ok, in that case we unfortunately can't verify if it's an old or new dps. I've ordered another DPS5005 from aliexpress, so we another unit to run ocd-client on, and to test should a fix be found.

kanflo, 2021-02-21:
@frederikvs For the original reverse engineering I used a hack saw to cut the pins to the display, soldered new ones and connected a logic analyser. 

I tried reaching out to the RD Official Store asking what has been changed. Let's keep our fingers crossed. To my knowledge there is only one 1.44" display to I am guessing that did not change.

daphreak66, 2021-02-21:
For reference, this is the the aliexpress link:

[Click](https://de.aliexpress.com/item/1005001304146754.html?spm=a2g0s.9042311.0.0.38244c4dfKfbbz)

frederikvs, 2021-03-03:
My new 3005 arrived, I've attached some dumps.
[dps_3005_v1.4.zip](https://github.com/kanflo/opendps/files/6075802/dps_3005_v1.4.zip)

I placed the same order at the same supplier, so I assume it's the same device, and will have the same issue if I flash it.
(For reference, I ordered the 3005 without communication from [here](https://www.aliexpress.com/item/4001238042800.html).)

@kanflo do you see anything in the outputs that could explain the issue we're having?

frederikvs, 2021-03-13:
Under the motto "He that breaks a thing to find out what it is, has wasted 20 bucks' worth of hardware", I made a valiant attempt at avoiding the hacksaw method :-) 

[image]

This means I can sniff on any of my 3 devices (openDPS 50V5A, 3005 stock firmware, and 3005 openDPS). After a few evenings of playing around with a scope and some python scripts (based on ripyl), I have some sniffs to share.

In the attached files, each line is a byte getting sent, and between brackets is L or H, depending on whether the D/CX signal was Low or High. I've compared the start of the openDPS sniff to the chip_init function in ili9163.c, and they match, so I'm reasonably confident that the sniffs are valid.

[opendps.txt](https://github.com/kanflo/opendps/files/6134763/opendps.txt)
[stock.txt](https://github.com/kanflo/opendps/files/6134764/stock.txt)

It's obvious that openDPS and the stock firmware have a different order to the commands they send. But at the same time, as far as I can tell, it's probably the same protocol that's getting sent over the wire.

(Minor note : in the sniff of the stock firmware, there's an (L)0x11 missing at the start. It comes way before all the rest of the bytes, and I couldn't fit it all on the scope with a high enough sample rate.)

frederikvs, 2021-03-13:
Good news : I have a working display on my 3005 again :-)

After some more scope investigation, I noticed that there was a signal on the stock 3005 that was missing from the opendps one. I think this is the `TFT_CSN` signal that was already in the openDPS code for the 5015. Enabling that alone wasn't enough though. It seems that the exact ordering of rising/falling edges on that CSN signal and the D/CX signal matters. 

With the stock firmware, it goes a like this : 
* a byte transfer
* CSN-signal goes up
* CSN goes down again
*  D/CX switches (if needed). 

With the openDPS with CSN enabled, it went like this : 
* byte transfer
* CSN goes up
* D/CX switches
* only then CSN goes down.

So enabling CSN and swapping the order of those signals, gives me a working display again.

@daphreak66 could you try my branch at https://github.com/frederikvs/opendps/tree/fix_3005 and let us know if it works for you too?

daphreak66, 2021-03-14:
The fix works! Thanks, good job!

daphreak66, 2021-03-14:
P.S.: Maybe we should make sure that it still works with the older hardware that didn't have the problem?

frederikvs, 2021-03-14:
I've already tried it on my 50V5A, and that works just as well with or without this fix :-) 

My expectation is that for most devices, this will simply be a signal that they didn't have (or need) before, and that they'll be getting now.
However, with the 5015 it's slightly different : the signal was needed, and I've just changed the timing of it. Can't be sure of the outcome, and I don't have the hardware to test it on.

kanflo, 2021-03-14:
Great work! IIRC my original DPS5005 did not use the nCS pin but rather kept it grounded. Please submit a PR when you feel ready and I will approve it. Looking at your changes I would recommend placing the ```gpio_clear(TFT_CSN_PORT, TFT_CSN_PIN);``` around line 123. The nCS pin should go low during the SPI transfer.

At some point I guess spi_driver.c needs some cleanup as ```gpio_clear(GPIOB, GPIO12);``` made me think I was sloppy reviewing a PR until I did git blame and learned it was my sin ;)

kanflo, 2021-03-14:
> P.S.: Maybe we should make sure that it still works with the older hardware that didn't have the problem?

Yes, I will try this on my ancient 5005.

frederikvs, 2021-03-14:
OK, I've found a better fix.

Adding a brief delay after the reset command is all it takes.

As it turns out, by enabling the `TFT_CSN` signal, and moving the `clear` to after the transfer, I had effectively disabled the first byte that gets tranferred. All other bytes are fine, because the nCS pin is still low from the previous transfer. Only that first `CMD_SWRESET` isn't, because the pin is initially set high in `hw.c`.

All sorts of wrong assumptions about edge ordering and the like, and a bit of serendipity have led us to a good solution in the end :-)
See #235

kanflo, 2021-03-14:
Good catch. Does this mean we can close this one?

frederikvs, 2021-03-14:
I think it can be closed :-)

closed 2021-03-14 by frederikvs

P208PUG, 2021-03-29:
> I think it can be closed :-)

Ohh any chance for precompiled files my aliexpress DPS5005 same issue white screen after flash , on windows is hard to get compile from source..

frederikvs, 2021-03-30:
[opendps_DPS5005.srec.zip](https://github.com/kanflo/opendps/files/6227368/opendps_DPS5005.srec.zip)

enjoy :-)

P208PUG, 2021-03-30:
#response_container_BBPPID{font-family: initial; font-size:initial; color: initial;}  The you save my life :)                                                                                                                                                                                                                                                                                                                         From: ***@***.***Sent: 30 March 2021 08:55To: ***@***.***Reply to: ***@***.***Cc: ***@***.***; ***@***.***Subject: Re: [kanflo/opendps] Blank screen on DPS3005 (#228)  
opendps_DPS5005.srec.zip
enjoy :-)