kanflo/opendps #231: White screen after flashing DPS5005

daphreak66, 2021-02-15:
Hi,

I got my DPS5005 from Aliexpress and flashed it according to the "Upgrading your DPS5005" instructions. The flash was successfull, as far as I can tell, both the firmware and the bootloader flash finished with "Verified OK" and "Resetting Target". I had to CTRL-C out because the script was stuck. I suppose this is due to the reset pin not being connected?

After the bootloader reset, the screen turns on, but only shows a solid white color, the device is not responsive. How can I help with this issue. I know my way around C/C++ code, but I have limited experience debugging firmware, since my main job is application development.

kanflo, 2021-02-15:
Did you flash app and then bootloader? You could try to get comfortable with ``arm-none-eabi-gdb``and see if the app is actually running.

daphreak66, 2021-02-16:
Hi. Yes, first the app, then the bootloader. Only after flashing the bootloader the screen turns white, without it the screen stays black. Something seems to be running, if I issue the "reset halt" command, the screen turns black again.

I'll ask the guys from our electronics department to set me up with gdb.

frederikvs, 2021-02-19:
This sounds very similar to what I'm seeing with my '3005, see #228.

@daphreak66  did you happen to capture the output of ocd-client.py before flashing?

daphreak66, 2021-02-19:
Unfortunately now. I wasn't sure if it's the same, since my screen at least turns white.

With some help I was able to run a debug session, but strange things happen. main() isn't even called, instead other random functions, not always the same. One time it was the reset_handler().

I've also tried flashing with st-flash and st-util instead of openocd, but the result is the same. Is it possible to download verified working binaries from somewhere, just to rule out the build process as an error source?

frederikvs, 2021-02-19:
My 3005 screen is also white - backlight on, but nothing displayed.

You could try with the files in attachment - they're compiled for the 5005, and I'm running them on my 50V5A without a problem.

[opendps_compiled.zip](https://github.com/kanflo/opendps/files/6011132/opendps_compiled.zip)

daphreak66, 2021-02-19:
Thanks, I'll try them later and report back

daphreak66, 2021-02-19:
Also no luck with your binaries.

This is all I get from gdb:

```
(gdb) break dpsboot.c:main 
Breakpoint 4 at 0x8000b60: file dpsboot.c, line 274.
(gdb) run
The program being debugged has been started already.
Start it from the beginning? (y or n) y
Starting program: /home/daphreak/workspace/opendps/dpsboot/dpsboot.elf 

Program received signal SIGTRAP, Trace/breakpoint trap.
0x20001fdc in ?? ()
```

daphreak66, 2021-02-19:
Forget the comment above, I used gdb and not arm-none-eabi-gdb unintentionally.

daphreak66, 2021-02-19:
I finally got the break points working, I'm not even sure what exactly I did to make it work. I can confirm that the event_handler processes keypress events (breakpoint at opendps.c:964). I think it's the same as issue #228, so I'll close this issue.

closed 2021-02-19 by daphreak66