kanflo/opendps #232: wz5005e support possible?

bm16ton, 2021-02-16:
The wz5005e shown here with teardown and pics,
https://robojax.com/learn/arduino/?vid=robojax_WZ5005E_5A_buck_LCD
 at the very least looks similiar, but supports the same io (but diff modbus protocol),same specs and damn close model number as the dps5005. I tried to look up the chips found in pictures but i had no luck. Anypictures available of the chips for the dps5005 so I could at least compare those before deciding to more forward? The wz5005 looks to be identical just with builtin ac? 
https://www.amazon.com/s?k=wz5005&i=industrial&ref=nb_sb_noss_2
So possibly two more models. I cant find any docs on the wz5005e modbuss protocol its 3.3v ttl uart, the official windows only software only has using a ch341 as requirement. Dunno if that requirement applies to dps5005 or not. I built a sniffer and started saving for various functions but I dont currently know anything about modbus and woud prefer not to have to go down that rabbit hole if I dont need too.

dclogicsCH, 2021-02-26:
Could you manage to get the default windows software to communicate with the device? For me the windows software doesn't even send anything to the com port. I have a act LED on my USB to TTL module and no activity can be observed whene i press the connect button on the software. Can you paste your sniffed recordings here?

bm16ton, 2021-02-27:
im currently on the road but i will post my sniff, unfortunatly it uses a
single byte crc at end and im not that up on crc to know the math. i found
2 versions of the software both required a ch34x but only one worked and
only if i tried to connect stopped then tried reconnecting. hopefully be
home soon and post those.

bm16ton, 2021-02-27:
ok so i start by sending
aa 01 20 01 00 00 00 00
00 00 00 00 00 00 00 00
00 00 00 cc

not sure what that does but it likesit first, to see current settings I
send ;
AA 01 2B 00 00 00 00 00
00 00 00 00 00 00 00 00
00 00 00 D6

the reply
AA 01 2B   14 50        12 AB          05 78         00 04           00 00
       A0 42   02 4A     00 00       A6
          "ovp"      "ocp amps"     "curr volt"   "curr set amp"   out
amps?        ?     lvp    ?out amps?    ?41026?

I have better notes on that reply but cannot find them ATM, to set voltage
and amps IE5v

AA 01 2C 13 88 12 AB 01
f4 00 04 00 00 00 42 00
00 00 00 6a

 but as i said Im unaware of the math for 1byte crc, convert to ascii add
then divide by 255 rings sum distant memory bell from the 90s. I didnt
bother with libs i straight up rednecked it by simply putting the hex in
files and using
xxd -r -p file > /dev/ttyUSB0 . The first bit I send I usualy send at least
twice to get reply then all isgood

kanflo, 2021-03-14:
Looks like a copy of the good old DPS so support should be quite easy for anyone willing to possibly sacrifice their WZ.

dclogicsCH, 2021-03-15:
So i'll flash my WZ with your firmware over the UART bootloader...

bm16ton, 2021-03-15:
i was curious how u get a locked/write protected firmware. ive seen
references people had made in passing like it was ez. I tried a handful of
apps etc but non of them gave me a 100% copy. i couldnt easily get at the
stm to and try and force the boot pins. And only swdio and swclk (vcc gnd)
where exposed. But id love too give it to yeah and learn the process

KaDw, 2021-12-01:
@dclogicsCH @bm16ton  any updates? Did you manage to run it on WZ5005e?

bm16ton, 2021-12-01:
I ended up boaring thru it like a true redneck.and simply cobled together
sum.basic shell scripts to control/read the device with stock
firmware...but god id love to know how to rip the firmware from a rdp1
device. I know its possible and one way is via loading an app into ram that
copies everything over say serial. If things turn out that u end up wanting
to stick with stock firmware i can get u started with the commands/scripts
basic protocol breakdown and crc. I have seen that at least 2 versions of
wz5005 exist. One (mine) only has one io option in settings (a
serial/modbus frankenstein) and the other has 3 i dont remember all the
options but it supports a esp8266 interface. I also use a esp on mine so no
loss, but the only person ive talked to who has the 3io options as soon as
they start serial comms it disables his buttons on the device. Mine dont
care i can use both. I noticed from pictures alone that the 3io devices
almost always have the psu built-in, mines just the control ruffly 3inches
by like 1.blah. which device do you have? Im planning on writing a simple
gui for the scripts (probly use gtkdialog) or simply do the majority
onboard an esp, thatd have the added bennefit of simply sending ascii to
the esp and it sending binary/hex without the added steps of passing
through on the linux side or have a website etc.

KaDw, 2021-12-01:
@bm16ton I just ordered WZ3605E from ali, it will take a while before I will put my hands on it.

bm16ton, 2021-12-01:
Thats awesome, the one i ordered came from uhg drok? I think i got that
right. And it didnt have a model number anywhere on it or in paperwork but
is an actual wz5005e. Its my first bench supply so not that familiar with
them is that model number the same or have sum cool bennefits? If i can ill
try and upload my code for controlling them and maybe a quick amd dirty
readme on my github tonight or tomorrow and post a link to it here. Just a
quick possibly useful tip, the "security" ha of the official software is
that it needs to be a ch340/ch341 usb to serial adapter to use it. I had
plenty but for the day i posted code i also grab stm32f103 firmwares that
mimick the ch34x whenever i see them so if anyone doesnt have one they have
at least another option. Ill throw them on the git as well

ltwin8, 2023-08-25:
Any updates on WZ3605E support?

it’s a nice compact Buck Boost module, Pitty it has such bad software.

bm16ton, 2023-08-25:
I've found the software to be not as bad as originally thought. I have sum
hacked up linux scripts to do basics, tho currently mostly hardcoded
values. I did port/modify/rewrite a esp8266 sketch oroginally for the drok
systems to it. Everything works except the on/off button randomly switches
random amounts of times almost like physical switches do with debounce. You
never know if its gonna land on on or off. Couldn't figure it out and moved
on. If I think of it when I get home I can  post the scripts and esp8266
firmware on github. Notes are spreadout thru various files and not
comprehensive but if your a linux guy with sum experience shouldnt be to
hard to figure out how to use them. Just an fyi multiple firmwares exist
for them. Mine has no native wifi support for example and just has a wanna
be modbus like interface that doesn't actually fit any specific protocol
I'm aware of. The other ones I've never used only seen in pics etc and may
operate differently. The crc if memory serves is a simple addem up and use
the last 2 digits. If I ever figure out the software debounce thing on the
web interface the rest did work based on my very limited testing. As for
the scripts always had plans on writing a simple front end with gtkdialog
but time is always a problem. If your using windows sum closed source
example from manufacturer exist. I downloaded a bunch and it seems two
versions exist with same version number and look identical. One doesnt work
at all the other required starting which would fail then stop then try
starting again and it woukd work. They require a ch340 or ch341 to operate.
Man I should get back on that sometime soon.ill reply here if I get those
posted and your interested. Oh almost forgot a german group wrote a
firmware for the wz's and droks (firmware for both) and userland utilities
as well but its closed source (or was) and I believe they only mentioned
the wz model with builtin psu and wifi etc support. Its been a while since
I checked on them so maybe have more info if it'd work on the smaller ones
or not. Unfortunatly don't remember any names.

g6094199, 2025-02-05:
> The wz5005e shown here with teardown and pics....

the same for the WZ6012

bm16ton, 2025-02-15:
So i had a 2nd wz5005 kicking around and just compiled opendps and flashed..boots to white screen :(  So ill hafta check its pinout vs the opendps. It will be a while but ill drop a line here if I get anywhere

g6094199, 2025-02-18:
since the WZ displays are much bigger i wont expect it to be working out of the box. can you access via modbus? this would show the mcu is working.

bm16ton, 2025-02-18:
modbus huh? Is the lcd really bigger? the wz5005e is only like 1.8inch, i
believe its bigger brother the wz5005 has a much larger screen. Yeah i
figured the init for the lcd probly hasta be changed (assuming its a
different lcd, seems likely) was also gonna check the pins being used,
unfortunately the stm32 is hidden under the lcd who's soldered to main pcb
via a header. To save me the soldering i figure ill just write a blinky
sketch and see if the currently used pins for the screen match up with
opendps pin defs. (using a multimeter) and take it from that. A quick run
thru of a bunch of init sequences after that should get me the lcd
controller type. Then just the pins for the power related pins?

g6094199, 2025-02-19:
ah ok. didnt know that the "e" version is a tiny one.
the original wifi adapter from sinilink just talks modbus on the serial lines. thats how its controlled via "app"