kanflo/opendps #238: Display is cut off at the bottom

FunnyGermanGuy, 2021-04-26:
Is there a possibility to change the display position? Because since I flashed opendps the input voltage and wifi symbol a cut off at the bottom.

kanflo, 2021-05-22:
What kind of DPS are you running?

FunnyGermanGuy, 2021-05-22:
The 5005 with the newer connectors

FunnyGermanGuy, 2021-05-25:
I thought maybe some more information and a picture would be useful. I bought an "DPS5005 + USB + BT" kit two month ago. It has the new smaller connectors and includes a bluetooth and a wifi expansion module. As seen in the photo, the "CV" text is still displayed in full height but the intake voltage ist cut at the bottom. The same goes for the wifi symbol when it's displayed. 

[image]

kanflo, 2021-05-26:
That's strange. If you dig some in the code you will find the coordinates for the offending widgets.

sercona, 2022-03-13:
same issue here; all my newer (march 2022) modules have the display cut off.  I edited my pull of the source and raised things by about 4 or 8 units and that did the trick.  wonder if some main x,y offsets have changed with some clone^(n) version of the tft display?