kanflo/opendps #239: What serial connector type does RuiDeng DPS5005 uses?

Harvie, 2021-05-21:
Hello,
i am sorry for off topic question, but i don't know where else to ask...

Do you know what kind of connector does RuiDeng DPS5005 use? I think it might be **1.25mm Molex PicoBlade**, but when i measure that connector using calipers, it seems fraction of milimeter different. Can anyone please confirm if that is 1.25mm PicoBlade or something else? Thank you!

[image]

kanflo, 2021-05-22:
The last time I checked it was a JST GH (ie. 1.25mm)

Harvie, 2021-05-23:
@kanflo Do you have same connectors as these? I made this photo of serial modules and my DPS5005 has exactly the same:

[image]

JST GH seems to look different...

kanflo, 2022-04-11:
A bit late perhaps, but yes. I have the same cables.

closed 2022-04-11 by kanflo