kanflo/opendps #249: Future of this project

danielkucera, 2022-03-19:
I was wondering if you still have time to maintain this project.
There seems to be several forks (@art103, @kohrar ,..) which are moving it forward by it's own way.

Maybe it would be better to unite, choose a new maintainer and keep all the development there. I see some new features there, I would like to add my own but cannot decide where to contribute.

What do you think?

lolyinseo, 2022-03-21:
It seems OpenDPS is no longer relevant(for example no DPS8005 support). There are other firmware that are actively developed. for example https://profimaxblog.ru/whats-new-in-firmware-4-0/

danielkucera, 2022-03-21:
> It seems OpenDPS is no longer relevant(for example no DPS8005 support). There are other firmware that are actively developed. for example https://profimaxblog.ru/whats-new-in-firmware-4-0/

Thanks, but that doesn't look like opensource to me...

sercona, 2022-03-21:
the profimax fw looks interesting, but since there is opensource fw, *that we can change and customize* - other close source fw is much much less interesting to me, at least.

I dont think this is at all 'irrelevant'.  I'm sorry I didn't see this years ago when it first came out ;)

kohrar, 2022-03-21:
That firmware has some interesting features. For me, OpenDPS is _good enough_ for the few DPS modules that I still have. I don't intend on getting more of these units and I haven't been keeping up with any of the newer offerings. 

However, it would be nice to see these pull requests get merged in. If I remember correctly, memory is sort of a constraining factor at the moment, so some effort would be required to make everything fit or have the make file include/exclude features cleanly.

kanflo, 2022-04-11:
Well, sadly I do not have much time to maintain OpenDPS anymore. If anyone creates a fork for maintenance I would be happy to refer to it.

violavon, 2023-02-17:
Are you still interested in the project?

@Xenoamor, @kohrar, @frederikvs, @tzarc, @danielkucera, @lukedempsey, @polihedron, @JackNewman12, @jimmyw, @goosenphil, @gojimmypi, @MarcusLindemann, @martinstep, @ilium007, @TheAssassin, @X-Ryl669, @m-kozlowski, @mabels, @Cabalist, @atx, @jenda122, @Spudmn, @cortices, @bjerrep

kohrar, 2023-02-17:
There's no need to spam everyone. As kanflo said, we don't have much time. You're free to fork and take on this project.

violavon, 2023-02-17:
@kohrar Sorry, I didn't mean any harm... Just sad about this project, it's the only open-source digital power supply firmware in the world.

violavon, 2023-02-17:
Maybe we could try crowdfunding, like Patreon?

TheAssassin, 2023-02-17:
Still, with every new message, over 24 people, many completely unrelated to the project, receive messages. You *knew* this was going to happen.

~~I hope you learned your lesson.~~ While typing, you posted even more spam. This is considered extremely bad behavior in open-source communities. Just because you're unhappy with any detail (for whatever reason, maybe the firmware just works for everyone involved so far...) doesn't justify spam like this.

@kanflo please lock the issue or ban @violavon, either stops the spam.

violavon, 2023-02-17:
Sorry (really). I was stupid to tag so many.

kanflo, 2023-02-27:
I think @violavon is a "young padawan" (joined a week ago) who learnt a lesson. Will close the issue though.

sercona, 2023-11-07:
I have my own fork and I am going to post it to my own github area, shortly.  it was a 'logn time ago' when I first did some mods and now that some time has gone by, I might have forgotten a few things.  but its still my go-to open source low-cost programmable psu.

justincr83, 2024-11-18:
hey, i know nothing about programming language or anything, but i have a question, is it possible to use a bugger screen on the original dps software?
thanks!

cortices, 2024-11-23:
@justincr83 
> hey, i know nothing about programming language or anything, but i have a question, is it possible to use a bugger screen on the original dps software? thanks!

it's not considered polite to post unrelated questions on an old thread like this -- it means every prior participant gets notifications.

But in answer, no, you absolutely cannot just use a bigger screen on this or any other DPS software.

closed 2025-01-29 by kanflo