kanflo/opendps #257: Feature request for Battery input power

TrashHauler, 2024-01-18:
Can you please add the following feature or guide me in the direction of coding it myself?

Request: Input/Battery minimum voltage cutout

Reason: Protect cordless tool battery as input power source from dropping below minimum safe voltage to prevent battery damage (Dewalt and Milwaukee batteries rely on the tool for low voltage cutout protection—Ryobi batteries have protection built into the battery)

Specifics: I haven’t done any programming in 30 years so I am more than a bit rusty. I would like to accomplish the following:
1) add a menu input and variable for “Battery Cutout Voltage” or maybe “input voltage cutout” default=zero
2) create battery / input voltage protection BVP function with the following basic logic
  If v_in_adc <= battery_cutout_voltage
    Set v_out and i_out = zero (handle_bvp?)
    Display on screen BVP activated at input voltage = battery_cutout_voltage
3) Check for BVP just like checking for OVP or OCP

Thanks in advance,

Bob

kanflo, 2024-02-15:
Sorry, this project is abandoned on my part. There was some activity in #249 in November last year so someone might still be developing for it.

closed 2024-02-15 by kanflo

sercona, 2024-12-27:
I like the idea of having a low voltage cutoff for battery input support.

I will try to add this to my own TODO list, as a fork of this project.