kanflo/opendps #260: 3d printable ryobi battery adapter for DPS

sercona, 2024-12-26:
some 3d printed ideas.  if you take the 'blank' battery adapter and add about 10mm to it (in slicer) you can fit the DPS5005 module in there with room to spare.  I removed the casing from my DPS and so I have discrete holes in the panel for all the openings needed.

[image]

[image]

[image]

[dps5005_ryobi_top_plate_stl.zip](https://github.com/user-attachments/files/18254889/dps5005_ryobi_top_plate_stl.zip)

kanflo, 2025-01-29:
Cool!

robots, 2026-01-29:
@sercona nice work, would it be possible to make the box fit DPS5020 board ?:)

sercona, 2026-01-29:
@robots the only one I designed was the zipfile that I posted here.  the printables ones are not from me.

the top plate that fits on the ryobi battery should fit any standalone module (not the split display and separate regulator pcb).  you have to remove the front bezel to fit the ryobi plate but then the 4 holes for screws should hold it in place from behind.