kanflo/opendps #263: DPS5005 Screen remains black, new hardware?

maribu, 2025-05-23:
Hi,

I tried to flash a new DPS5005 with OpenDPS, but the device seems to be bricked:

- screen remains black when powered
- `./dpsctl.py -d /dev/ttyUSB0 --ping` gives
	> `Error: timeout talking to device /dev/ttyUSB0.`

SWD access is still available.

<details><summary>Output of <code>./ocd-client.py all</code> when output is disabled</summary>

```
>>>> adc1
ADC1 settings
SR       : 0x00000010  [0x40012400]
CR1      : 0x00000000  [0x40012404]
CR2      : 0x009e0001  [0x40012408]
SMPR1    : 0x00000000  [0x4001240c]
SMPR2    : 0x36c00000  [0x40012410]
JOFR1    : 0x00000000  [0x40012414]
JOFR2    : 0x00000000  [0x40012418]
JOFR3    : 0x00000000  [0x4001241c]
JOFR4    : 0x00000000  [0x40012420]
HTR      : 0x00000fff  [0x40012424]
LTR      : 0x00000000  [0x40012428]
SQR1     : 0x00000000  [0x4001242c]
SQR2     : 0x00000000  [0x40012430]
SQR3     : 0x00000008  [0x40012434]
JSQR     : 0x00000000  [0x40012438]
JDR1     : 0x00000000  [0x4001243c]
JDR2     : 0x00000000  [0x40012440]
JDR3     : 0x00000000  [0x40012444]
JDR4     : 0x00000000  [0x40012448]
DR       : 0x0000013c  [0x4001244c]

>>>> afio
AFIO settings
EVCR     : 0x00000000  [0x40010000]
MAPR     : 0x02008000  [0x40010004]
EXTICR1  : 0x00000000  [0x40010008]
EXTICR2  : 0x00000011  [0x4001000c]
EXTICR3  : 0x00000011  [0x40010010]
EXTICR4  : 0x00000000  [0x40010014]
MAPR2    : 0x00000000  [0x40010018]

>>>> dac
DAC settings
CR       : 0x00030003  [0x40007400]
SWTRIGR  : 0x00000000  [0x40007404]
DHR12R1  : 0x00000000  [0x40007408]
DHR12L1  : 0x00000000  [0x4000740c]
DHR8R1   : 0x00000000  [0x40007410]
DHR12R2  : 0x00000000  [0x40007414]
DHR12L2  : 0x00000000  [0x40007418]
DHR8R2   : 0x00000000  [0x4000741c]
DHR12RD  : 0x00000000  [0x40007420]
DHR12LD  : 0x00000000  [0x40007424]
DHR8RD   : 0x00000000  [0x40007428]
DOR1     : 0x00000000  [0x4000742c]
DOR2     : 0x00000000  [0x40007430]
SR       : 0x00000000  [0x40007434]

>>>> dma
DMA settings
ISR      : 0x00000000  [0x40020000]
IFCR     : 0x00000000  [0x40020004]
CCR1     : 0x00000000  [0x40020008]
CNDTR    : 0x00000000  [0x40020084]
CPAR1    : 0x00000000  [0x40020010]
[... 322 lines cut ...]
CCER     : 0x00000010  [0x40000820]
CNT      : 0x000022ea  [0x40000824]
PSC      : 0x00000000  [0x40000828]
ARR      : 0x00005dbf  [0x4000082c]
RCR      : 0x00000000  [0x40000830]
CCR1     : 0x00000000  [0x40000834]
CCR2     : 0x00004e20  [0x40000838]
CCR3     : 0x00000000  [0x4000083c]
CCR4     : 0x00000000  [0x40000840]
BDTR     : 0x00000000  [0x40000844]
DCR      : 0x00000000  [0x40000848]
DMAR     : 0x00000081  [0x4000084c]
```

</details>

```
xPSR: 0x61000000 pc: 0x080006ec msp: 0x20001fb8
Info : device id = 0x10016420
Info : flash size = 64 KiB
```

Note that I tried both building from source (I had to use https://github.com/kanflo/opendps/pull/254 due to using `python2` not really being an option anymore), as well as the prebuild binaries (copied them into `opendps` and `dpsboot`, respectively, before running `make flash` in `opendps and `dpsboot`).

Am I doing something wrong, or is my hardware a newer revision that is not supported yet?

kanflo, 2025-05-25:
I should have merged #254 some time ago and saved you some time. Sorry about that.

maribu, 2025-05-25:
No need to apologize.

Thanks a lot for the great firmware! It so great to have a lab power supply that I can control from command line / python :)